ZJU-FAST-Lab/Fast-Drone-250 #2: Few Questions regarding the setup

FaboNo, 2021-11-03:
First thank you very much for providing all these information! It is very helpful for students and researchers.
I have few question

1. Can you tell us please what are the characteristics of the tri-blades you are using? I cannot find it in the shopping list - or I missed it ...
2. From what I can see the total weight of the drone fully assembled should be around 1.7/1.8 kg probably. I am wondering what is the thrust delivered by the T-motor KV 2550 at 50%/ 60% throttle ? I tried to find out on Internet but cannot... and the related question is about the autonomy - is it around 5 mn or more?
3. You are not mentioning what kind of geometric controller are you using?  Can you tell us more about it? 
4. As you are using PX4 and dynamic trajectories, it is pretty important to have the "PX4  rate controller"  pretty well tuned.  is it possible for you to share your params file ?

Thanks again for all the fantastic open you are sharing with the community

PanNeng-ZJU, 2021-11-09:
@FaboNo Hello, thanks for your advise.
1.Various types from GEMFAN work. For example, the 5147, 5144 tri-blades are fine.
2.The weight of the fully assembled drone is around 1.1kg in fact. You can look up the thrust-efficiency table at https://uav-cn.tmotor.com
3.We use the SO(3) px4 controller onboard. And for this part, I'm afraid we can't open-source it.
4.The params file and the corresponding firmware are already updated.

roam-water, 2021-11-15:
首先非常感谢你们提供的信息，请问一下，我组装的qav250机子跟您的重量也是11.1kg，但是我推油门时候总是上升或者下降很快（动力太大啦），非常难控制，请问您是怎么调试让无人机平稳上升，然后油门控制比较柔和，动作顺畅的呢，是修改PWM_MAIN_RATE 吗，还是做出了什么样的修改，期待你的回答，同时非常感谢您抽出时间阅读我的留言，祝您快乐！

PanNeng-ZJU, 2021-11-22:
@roam-water  如果你是用pwm控制电机的话，可以通过修改pwm_main_rate来解决这个问题，但这个教程里是用dshot控制电机的。可以通过地面站里调节飞机的悬停油门来解决这个问题。

closed 2021-11-22 by PanNeng-ZJU

ZhangBihui0311, 2022-02-25:
@PanNeng-ZJU 
请问能否解释一下“SO(3) px4 controller”，它和PX4发行的不同版本firmware有什么关系？
另外还想请问作者是否采用了PX4的offboard控制模式？我在自己的实验中感觉不同固件版本对offboard支持很不稳定。
谢谢！

roam-water, 2022-03-21:
谢谢，非常感谢