ZJU-FAST-Lab/Fast-Drone-250 #3: 有关可选购的飞控

J1dan, 2021-11-27:
你们好，请问可选购的三种飞控有什么使用上的区别吗？是否有更推荐的一种？

PanNeng-ZJU, 2021-11-29:
@J1dan 
1.个人更推荐CUAV NORA，因为体积相较其他两款更小一些，性能上都比较相似。
2.表中的载板体积更小，更适合使用在无人机上，但相应的USB接口也更少，如果您对机载电脑体积没有要求，可以不购买额外的载板。另外官方的底板不太好固定在飞机上，可能需要用比较厚的海绵胶粘起来

closed 2021-11-29 by PanNeng-ZJU

J1dan, 2021-11-29:
非常感谢您！

J1dan, 2021-11-30:
您好，再请问一下如果是购买NX载板的话，是不是就只需要购买NX核心板然后装上去就可以了？也就是说不需要购买NX开发板？我看价目表里的Xavier NX价格是6800，但核心板的价格只要3699，所以有点好奇如果另买载板的话是否还是需要购买开发者套件。

PanNeng-ZJU, 2021-11-30:
我的建议是不缺预算的话购买开发者套件。因为单独购买的核心板是不能插SD卡的，装系统的SD卡必须插在NX载板上，而这需要另打驱动，比较麻烦。开发者套件里的核心板是支持SD卡的，可以即插即用

抄送: "自动化（交叉）1701 ***@***.***&gt;;"State ***@***.***&gt;;

J1dan, 2021-11-30:
谢谢解答，但是我还是有点迷惑：之前您说表中的载板体积更小，我理解为更推荐使用表中的载板；但是如果购买开发板的话，不是就不需要另外的载板了吗？
或者换句话说，我现在不缺预算，那么是不是更推荐购买开发者套件加上表中889的载板？还是只需要开发者套件，或官方核心板+889载板？麻烦您建议一下。麻烦您了！

PanNeng-ZJU, 2021-11-30:
购买开发者套件+889的载板，把开发者套件里的核心板拆出来装在889的载板上

抄送: "自动化（交叉）1701 ***@***.***&gt;;"State ***@***.***&gt;;

J1dan, 2021-11-30:
谢谢，刚才去问了一下开发板的卖家，但他们说开发板的SD卡也是插在底板上的，所以单卖的核心板和开发板上的核心板应该没有区别；而889载板的卖家说可以在他们家买核心板+他们的底板，可以帮忙装好ubuntu系统，随后我们再自己装SD卡，根据他们的教程设置从SD卡启动系统。请问您觉得这个方案怎么样？

PanNeng-ZJU, 2021-11-30:
可以的

抄送: "自动化（交叉）1701 ***@***.***&gt;;"State ***@***.***&gt;;