ZJU-FAST-Lab/Fast-Drone-250 #5: 您好，想向您请教一下关于飞控和机载计算机的通信线路问题

yangy-98, 2022-03-14:
您好，想请教一下，请问您们在对飞控和机载计算机通信时，用的是什么设备线路呢？我看PX4官网推荐了mRo USB FTDI Serial to JST-GH (Basic)，但似乎买不到这样的线路。
希望能收到您的回复，感谢！

PanNeng-ZJU, 2022-03-14:
用普通的能传输数据的usb线就可以，例如cuav的飞控一般是typec口的，就用typec-usbA的线就行

closed 2022-03-14 by PanNeng-ZJU

yangy-98, 2022-03-14:
感谢您的回复！

还想再请教您一下，目前我在用的飞控和机载计算机是pixhawk+nuc，我找了一下它们之间的通信方式，网络上似乎都是将机载计算机用usb转串口连接pixhawk上的telem2端口来进行的通信，

我之前也关注到您有用nuc主机的打算，不知您是否了解pixhawk+nuc的通信方式，以上所述是唯一方式？还是您所说的用普通的能传输数据的usb线就可以？

再次期待能够收到您的回复，感谢！

yangy-98, 2022-03-20:
> 用普通的能传输数据的usb线就可以，例如cuav的飞控一般是typec口的，就用typec-usbA的线就行

感谢您的回复！

还想再请教您一下，

1. 目前我在用的飞控和机载计算机是pixhawk+nuc，我找了一下它们之间的通信方式，网络上似乎都是将机载计算机用usb转串口连接pixhawk上的telem2端口来进行的通信，我之前也关注到您有用nuc主机的打算，不知您是否了解pixhawk+nuc的通信方式，以上所述是唯一方式？还是您所说的用普通的能传输数据的usb线就可以？

2. 请问您们是如何解决飞控和机载计算机的供电问题的？是用一块航模电池通过分电板的方式给它们供电的吗？

期待能够再次收到您的回复，感谢！

YU-TON, 2022-04-07:
1.用telem2需要usb转ttl的线连起来，我之前就是这么连的，也可以飞控的micro usb连到计算机的usb
2.教程里供电就是直接插电源
我这有一个疑问，直接插电源的话会不会在飞机降落的时候有大电流对计算机造成损坏，因为我之前有一架飞机的jetson NX是连到一个有过电流保护的12v稳压模块，每次降落螺旋桨停转瞬间计算机就会自动关机，我想是因为电压或者电流波动幅度过打导致稳压模块自动断电，那么如果不接稳压，这个波动是不是会对计算机有较大的损坏呢
@yangy-98 @PanNeng-ZJU

yangy-98, 2022-04-07:
非常感谢您们二位的回复，
关于稳压模块我暂时还没有用上，我目前使用的似乎是旧版的nuc，光供电电压好像就需12v了，目前还在初步测试中。
@PanNeng-ZJU @YU-TON