ZJU-FAST-Lab/Fast-Drone-250 #6: heading issue

skyaid, 2022-04-07:
Hello, thank you for a great project.

When switching the current Px4ctrl to offboard, it rotates 180 degrees.

We start by looking at the back of the aircraft, but if we change to offboard

The aircraft sees us. 

When looking at vio and ego planer rviz, the initial position is correct.

PanNeng-ZJU, 2022-04-07:
What is the odometry you use? Did you check the odometry's coordinate system si correct?

skyaid, 2022-04-07:
I used vins fusion gpu and odometry coodinate position is correct

I replaced the estimator file and visualization file using vins fusion provided by zju

PanNeng-ZJU, 2022-04-07:
Sorry, I have no idea where the problem lies @skyaid

skyaid, 2022-04-08:
@PanNeng-ZJU thanks

closed 2022-04-08 by skyaid