ZJU-FAST-Lab/Fast-Drone-250 #7: 运行ego-planner出现的问题

SSZ1, 2022-04-14:
您好！我一直都在关注你们组。首先非常感谢您能够开源自主无人机，可以预见到一定能够带动国内专业领域的发展。
     我在NUC上配置ego-planner这一步时，按照你们给的命令，编译完成，并且执行了`source devel/setup.bash`，在执行`roslaunch ego_planner sing_run_in_sim.launch`时报错：

**RLException: [sing_run_in_sim.launch] is neither a launch file in package [ego_planner] nor is [ego_planner] a launch file name
The traceback for the exception was written to the log file.**

     请问为什么会出现这个问题，该怎么解决呢？十分感谢
[image]

xzqiaochu, 2022-04-16:
您好，似乎是拼写错误：single误拼成了sing
正确的写法：roslaunch ego_planner single_run_in_sim.launch
如果解决了记得close一下哈

SSZ1, 2022-04-17:
> 

谢谢您！已经解决了，之前好像是看了没更新的readme.md，现在已经更新了，非常感谢！！！

closed 2022-04-17 by SSZ1