ZJU-FAST-Lab/Fast-Drone-250 #13: CUAV X7+飞控

meyiao, 2022-06-07:
您好，买了purchase_list里边链接的x7飞控，买到才发现其实是x7+。烧不进去给的固件，请问用px4 v1.12.3以后的固件可以吗？有哪里要修改的吗？
谢谢!

albert17github, 2022-07-10:
您好，我使用的也是CUAV x7+飞控，希望对您有点帮助。
首先，这个项目文件下的飞控固件是给V5使用的，x7+不能使用。
如果要用x7+飞控，得自己再下载PX4的固件：
PX4固件下载链接： https://github.com/PX4/PX4-Autopilot/releases
试了一下，使用v1.13的固件时，imu的频率有点问题。
我用的是**x7pro v1.11**的固件，目前来看，没有遇到什么问题。

albert17github, 2022-07-10:
> 您好，买了purchase_list里边链接的x7飞控，买到才发现其实是x7+。烧不进去给的固件，请问用px4 v1.12.3以后的固件可以吗？有哪里要修改的吗？ 谢谢!

您好，我使用的也是CUAV x7+飞控，希望对您有点帮助。
首先，这个项目文件下的飞控固件是给V5使用的，x7+不能使用。
如果要用x7+飞控，得自己再下载PX4的固件：
PX4固件下载链接： https://github.com/PX4/PX4-Autopilot/releases
试了一下，使用v1.13的固件时，imu的频率有点问题。
我用的是x7pro v1.11的固件，目前来看，没有遇到什么问题。

meyiao, 2022-07-11:
> > 您好，买了purchase_list里边链接的x7飞控，买到才发现其实是x7+。烧不进去给的固件，请问用px4 v1.12.3以后的固件可以吗？有哪里要修改的吗？ 谢谢!
> 
> 您好，我使用的也是CUAV x7+飞控，希望对您有点帮助。 首先，这个项目文件下的飞控固件是给V5使用的，x7+不能使用。 如果要用x7+飞控，得自己再下载PX4的固件： PX4固件下载链接： https://github.com/PX4/PX4-Autopilot/releases 试了一下，使用v1.13的固件时，imu的频率有点问题。 我用的是x7pro v1.11的固件，目前来看，没有遇到什么问题。

感谢大佬，我有时间试试！

closed 2022-07-11 by meyiao