ZJU-FAST-Lab/Fast-Drone-250 #27: sh rspx4.sh 一段时间后出现wait for imu

cbyf, 2022-10-31:
用rostopic hz /mavros/imu/data_raw 检查频率为190hz（刚刚开始时会从150hz逐渐上升）。拿起飞机来回走的过程中会出现**wait for im**u。
问题如图
[image]

PanNeng-ZJU, 2022-11-09:
1. wait for imu是因为imu预积分和特征点匹配间相差太远，说明已经飘了

2. 右手系 x前 y左 z上

closed 2022-11-09 by PanNeng-ZJU

cbyf, 2022-11-12:
请问如何改善imu预积分和特征点匹配间相差太远的问题呢？谢谢

PanNeng-ZJU, 2022-11-14:
@cbyf 参考Q&A中的 “VINS飘怎么办？”

qingbyin, 2023-03-01:
Hi @PanNeng-ZJU,
看到这个回复有些疑惑，请教下无人机IMU的坐标系通常是[NED](https://docs.px4.io/main/en/config/flight_controller_orientation.html) ，为什么这里机体坐标系是前左上呢?并且这样约定的话，那IMU也要重新设定为z朝上了?

70loKirin, 2023-03-12:
你好，请问你解决了吗，最近我也在尝试，跟你一样出现过wait for imu的情况，我目前最大的疑点是旋转矩阵的填法，因为视频中飞控和相机有一个90度的差值，但是在我的飞机上二者都朝前
请问你知道这种情况下旋转矩阵怎么填吗
以及我也想知道原理，即imu和相机的坐标系是怎样的，我到现在还没有根据他给出的这个矩阵反推出二者的关系，网上的资料说法太多了（比如相机是否是右x下y前z 或是普通的前x左y右z   imu的方向也是 ned还是普通的情况？）
打扰抱歉 但这真的很困扰我

bulingda, 2023-04-04:
> 你好，请问你解决了吗，最近我也在尝试，跟你一样出现过wait for imu的情况，我目前最大的疑点是旋转矩阵的填法，因为视频中飞控和相机有一个90度的差值，但是在我的飞机上二者都朝前 请问你知道这种情况下旋转矩阵怎么填吗 以及我也想知道原理，即imu和相机的坐标系是怎样的，我到现在还没有根据他给出的这个矩阵反推出二者的关系，网上的资料说法太多了（比如相机是否是右x下y前z 或是普通的前x左y右z imu的方向也是 ned还是普通的情况？） 打扰抱歉 但这真的很困扰我

请问你解决了吗？

qllokirin, 2023-04-04:
> > 你好，请问你解决了吗，最近我也在尝试，跟你一样出现过wait for imu的情况，我目前最大的疑点是旋转矩阵的填法，因为视频中飞控和相机有一个90度的差值，但是在我的飞机上二者都朝前 请问你知道这种情况下旋转矩阵怎么填吗 以及我也想知道原理，即imu和相机的坐标系是怎样的，我到现在还没有根据他给出的这个矩阵反推出二者的关系，网上的资料说法太多了（比如相机是否是右x下y前z 或是普通的前x左y右z imu的方向也是 ned还是普通的情况？） 打扰抱歉 但这真的很困扰我
> 
> 请问你解决了吗？

嗯，忘记跟进了，我认为主要问题就是齐次变换矩阵中的旋转矩阵
简而言之，使用px4的imu和d435i前视就直接按照视频里的填即
[0 0 1
-1 0 0
0 -1 0]（无论飞控朝向哪边安装，飞控会在校准的时候知晓前方的位置，最终发布的imu都是一样的 
但括号内的内容我没有测试过 只是推测）
（相机坐标系右x下y前z  px4的imu是前x左y上z 自行计算一下 旋转矩阵就是上面那个）
然后如果没有光源频闪（rqt_image_view仔细看看）且特征点足够，就不会出现了，多次标定把矩阵填准就好了