ZJU-FAST-Lab/Fast-Drone-250 #35: pixhawk4 no longer available

FaboNo, 2022-11-29:
thank you very much for sharing this amazing project (and years of research) with the community
I see that pixhawk4 is sold out almost everywhere so I am wondering if there is any issue with the latest version of pixhawk 6C or CUAV? Indeed in this case, one have to use a difference release of PX4 such px4 fmu v6 for instance

PanNeng-ZJU, 2022-11-30:
We're sorry for this. CUAV V5+works well, and some of the students of this course have tried pixhawk 6C and it turns out fine.

closed 2022-11-30 by PanNeng-ZJU

EhrazImam, 2022-11-30:
Hi @PanNeng-ZJU well i tried with cube orange and it works fine as i am also having pixhawk 6C but as soon as i am turning of the magnetometer(as given in the README) i am not getting the /mavros/imu/data. As i can see you mention you have tried pixhawk 6C cane you you just help me with same.
Thankyou

PanNeng-ZJU, 2022-11-30:
@EhrazImam I'm sorry that I didn't try pixhawk 6C on my own, but some of the students did. I guess you can try different version of firmware, like 1.13, and it may work.

EhrazImam, 2022-11-30:
@PanNeng-ZJU I tried it with 1.13.0 master version, 1.13.1, 1.13.2 and i found the same error where i cube orange is working fine in all version.

PanNeng-ZJU, 2022-11-30:
@EhrazImam In that case, I think you can refer to the holybro manufacturer or px4 official for more advice.

EhrazImam, 2022-11-30:
@PanNeng-ZJU Thankyou, yeah i have been trying to ask them but i couldn't find any reply from them, it will be great if i could connect with the students that have tried on pixhawk 6C.

FaboNo, 2022-11-30:
@EhrazImam the orange cube is a pixhawk2 version, you were able to run px fmu v5 on it? I have the black one so I will give it a try
@PanNeng-ZJU I did not see in the README that you need to deactivate the magnetometer - can you tell us why is it necessary?
Thank you very much

EhrazImam, 2022-11-30:
@FaboNo No havent tried with FMU v5 same out of stock so haven't tried

BruceShine, 2022-11-30:
I have successfully run ego-planner with pixhawk 6c. Have you tried daily build QGroundControl to burn firmware?
https://docs.qgroundcontrol.com/master/en/releases/daily_builds.html

EhrazImam, 2022-11-30:
> I have successfully run ego-planner with pixhawk 6c. Have you tried daily build QGroundControl to burn firmware? https://docs.qgroundcontrol.com/master/en/releases/daily_builds.html

Hi yeah i have tried did you change the SYS_MC_EST_GROUP = Q atttitude or you are running it without changing this.

EhrazImam, 2022-12-09:
Hi @BruceShine cane you please clear this issue

BruceShine, 2022-12-14:
> > I have successfully run ego-planner with pixhawk 6c. Have you tried daily build QGroundControl to burn firmware? https://docs.qgroundcontrol.com/master/en/releases/daily_builds.html
> 
> Hi yeah i have tried did you change the SYS_MC_EST_GROUP = Q atttitude or you are running it without changing this. And which firmware version you are using. Thankyou

I did not change that parameter. As for firmware, just use the recommended stable version by QGC.

EhrazImam, 2022-12-30:
@FaboNo @BruceShine @PanNeng-ZJU  I am facing running VINS FUSION with Pixhawk 6C as i have run it with Cube Orange and Pixhawk 4 and D435i's IMU and it was running very smoothly.
However i am not sure this is due to Pixhawk 6C's IMU please have a look at IMU.yaml(calibrated with imu_utils).

#Accelerometers
accelerometer_noise_density: 8.7630907909861493e-03   #Noise density (continuous-time)
accelerometer_random_walk:   3.6846874274508294e-04   #Bias random walk

#Gyroscopes
gyroscope_noise_density:     8.9757009701970693e-04  #Noise density (continuous-time)
gyroscope_random_walk:       7.5769383737709901e-06   #Bias random walk

rostopic:                    /mavros/imu/data      #the IMU ROS topic
update_rate:                 200.0     #Hz (for discretization of the values above)

Calibration results
===================
Normalized Residuals
----------------------------
Reprojection error (cam0):     mean 0.3304897901167408, median 0.2200004172986265, std: 0.41110001871708324
Reprojection error (cam1):     mean 0.32047461695261253, median 0.210521224108912, std: 0.40733175830316104
Gyroscope error (imu0):        mean 2.3650272386308853, median 0.900469841552671, std: 3.6080615076365934
Accelerometer error (imu0):    mean 0.8969253003256608, median 0.5084065251567538, std: 1.3295349711772715

Residuals
----------------------------
Reprojection error (cam0) [px]:     mean 0.3304897901167408, median 0.2200004172986265, std: 0.41110001871708324
Reprojection error (cam1) [px]:     mean 0.32047461695261253, median 0.210521224108912, std: 0.40733175830316104
Gyroscope error (imu0) [rad/s]:     mean 0.031485881950145136, median 0.011988059447131917, std: 0.04803454135440456
Accelerometer error (imu0) [m/s^2]: mean 0.11658023498803086, median 0.06608148097807301, std: 0.17280981962307645

Transformation (cam0):
-----------------------
T_ci:  (imu0 to cam0): 
[[-0.0011831  -0.99949408 -0.03178334  0.02789416]
 [-0.01458597  0.03179723 -0.99938791  0.00909488]
 [ 0.99989292 -0.00071878 -0.01461621 -0.06212618]
 [ 0.          0.          0.          1.        ]]

T_ic:  (cam0 to imu0): 
[[-0.0011831  -0.01458597  0.99989292  0.06228519]
 [-0.99949408  0.03179723 -0.00071878  0.0275462 ]
 [-0.03178334 -0.99938791 -0.01461621  0.00906783]
 [ 0.          0.          0.          1.        ]]

timeshift cam0 to imu0: [s] (t_imu = t_cam + shift)
-0.0016585079261203777

Transformation (cam1):
-----------------------
T_ci:  (imu0 to cam1): 
[[-0.0045073  -0.99947521 -0.03207801 -0.02228279]
 [-0.0139655   0.03213812 -0.99938586  0.0090407 ]
 [ 0.99989232 -0.00405655 -0.01410303 -0.06163653]
 [ 0.          0.          0.          1.        ]]

T_ic:  (cam1 to imu0): 
[[-0.0045073  -0.0139655   0.99989232  0.06165571]
 [-0.99947521  0.03213812 -0.00405655 -0.02281168]
 [-0.03207801 -0.99938586 -0.01410303  0.0074511 ]
 [ 0.          0.          0.          1.        ]]

timeshift cam1 to imu0: [s] (t_imu = t_cam + shift)
-0.0017732647034794681

Baselines:
----------
Baseline (cam0 to cam1): 
[[ 0.99999443  0.00034357 -0.00331956 -0.05038615]
 [-0.00034152  0.99999975  0.00062013 -0.00000612]
 [ 0.00331977 -0.00061899  0.9999943   0.00040233]
 [ 0.          0.          0.          1.        ]]
baseline norm:  0.05038775847016609 [m]

Gravity vector in target coords: [m/s^2]
[ 9.80627198 -0.01036926  0.07311062]

Calibration configuration
=========================

cam0
-----
  Camera model: pinhole
  Focal length: [370.7108040376509, 372.99870767379053]
  Principal point: [319.78661509710616, 239.45619458177592]
  Distortion model: radtan
  Distortion coefficients: [-0.007247130905233888, -0.004773953477251725, -0.0018260889423888547, -0.00011655993921176293]
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.022 [m]
    Spacing 0.006599999999999999 [m]

cam1
-----
  Camera model: pinhole
  Focal length: [370.7135602083122, 373.1960137433778]
  Principal point: [321.2458053488056, 239.32180866261118]
  Distortion model: radtan
  Distortion coefficients: [-0.00549041557346843, -0.004442582242381625, -0.002207008976176078, -9.093312682821986e-06]
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.022 [m]
    Spacing 0.006599999999999999 [m]

IMU configuration
=================

IMU0:
 ----------------------------
  Model: calibrated
  Update rate: 220.0
  Accelerometer:
    Noise density: 0.00876309079098615 
    Noise density (discrete): 0.12997764133278683 
    Random walk: 0.00036846874274508294
  Gyroscope:
    Noise density: 0.0008975700970197069
    Noise density (discrete): 0.013313115991159715 
    Random walk: 7.57693837377099e-06
  T_i_b
    [[1. 0. 0. 0.]
     [0. 1. 0. 0.]
     [0. 0. 1. 0.]
     [0. 0. 0. 1.]]
  time offset with respect to IMU0: 0.0 [s]

As soon i am taking yaw it's drifting a lot.
Thankyou