ZJU-FAST-Lab/Fast-Drone-250 #36: 关于控制器部分的问题

pugummy, 2022-12-05:
您好！首先感谢您的分享！
请问px4ctrl里面控制器设计部分有没有参考文献呢？不是很明白控制器输出的姿态的计算方法。^_^

dongdongbh, 2023-01-13:
同疑问，不是很理解这两行

https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/8ff64274b161bb829653faf5946656658e7e2c9f/src/realflight_modules/px4ctrl/src/controller.cpp#L41-L42

是一种参考重力向量（认为机体加速度较小）对姿态的粗略估算吗？

FaboNo, 2023-01-31:
@dongdongbh the equations are coming from which paper?
方程式来自哪篇论文？

Nikanbaturu, 2023-02-15:
@dongdongbh 您好！请问可以把controller理解为位置控制器吗？而且没太看懂对debug_msg_赋值的意义，是为了存储计算得到的期望值将其作为内级控制器的输入吗？

dongdongbh, 2023-02-15:
debug_msg_只是为了debug方便而发布的topic，最终油门和姿态给了飞控，控制是位置和速度的P控制外加加速度前馈给到姿态控制器做串级

Nikanbaturu, 2023-02-16:
@dongdongbh 感谢您的回答，那controller这部分的位置控制算法是否可以替换成其他的呢？

dongdongbh, 2023-02-16:
可以

dongdongbh, 2023-02-22:
[image]
Finally understand it by reading their PPT

FaboNo, 2023-02-22:
@dongdongbh oh very interesting, where did you find the ppt?

dongdongbh, 2023-02-23:
Here is a [video](https://www.bilibili.com/video/BV1Jq4y1T7QD/?spm_id_from=333.337.search-card.all.click) with this ppt talking about these things, but in Chinese.

karjalv, 2023-02-27:
> 另外关于Recursive least squares algorithm with vanishing memory
> 
> https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/8ff64274b161bb829653faf5946656658e7e2c9f/src/realflight_modules/px4ctrl/src/controller.cpp#L135
> 
> 为什么要除 `rho2_`，原始的Recursive least squares algorithm这里似乎不需要除
> 
> [image]

@dongdongbh Have you found any reason for this division with `rho2_` ? I've been wondering the same thing.

bulingda, 2023-03-24:
> @dongdongbh Have you found any reason for this division with `rho2_` ? I've been wondering the same thing.

Hello, do you understand the meaning of this formula?

closed 2023-04-12 by PanNeng-ZJU