ZJU-FAST-Lab/Fast-Drone-250 #39: Few questions about the firmware release and Pixhawk orientation

FaboNo, 2023-01-23:
Dear All,

As mentioned in a previous message, instead of a pixhawk 4 mini (no more available), I had to purchase a Pixhawk 6x mini.
In your READ.ME you mentioned that we have to use the v1.11.0 but this version does not support the Pixhawk 6x (starting with v1.12.0). So the question is can you tell us why the v1.13 is not suitable, and is it still not suitable with the new hardware?

Regarding the assembling, we have to set up the pixhawk "perpendicular" to the front face of the drone. Why do we have to do this? Because of VINS? Because when we are testing the drone with the pixhawk only and the AT9S PRO, if the pilot wants to go forward, the drone will move sideway
[image]

Thank you very much for answering the question and helping us to reproduce what you do in your lab!

PanNeng-ZJU, 2023-04-12:
1. we were having some trouble of unstable measurement of VIO with the 1.13.0 firmware, but I think you can give it a try in the pixhawk 6c.
2. The reason why we set up the pixhawk "perpendicular" to the front face of the drone, is that it's more suitable for installation. As long as the orientation of the flight controller is configured properly in the calibration part in the QGC, you can install the flight controller any way you want.

closed 2023-04-12 by PanNeng-ZJU

EhrazImam, 2023-08-30:
@FaboNo hope you are doing well were you able to run on pixhawk 6x

EhrazImam, 2023-08-30:
Any other hardware you have used

For 200 hz create a file etc/extras.txt and add  mavlink stream -d /dev/ttyS0 -s HIGHRES_IMU -r 200 
mavlink stream -d /dev/ttyS0 -s ODOMETRY -r 200

FaboNo, 2023-09-03:
@EhrazImam yes it is working by doing this (I saw it in on issue):
roslaunch mavros px4.launch &sleep 10;
rosrun mavros mavcmd long 511 105 4550 0 0 0 0 0 &sleep 1;
echo "/mavros/imu/data_raw to 200Hz"
rosrun mavros mavcmd long 511 31  4550 0 0 0 0 0 &sleep 1;
echo "/mavros/imu/data to 200Hz"