ZJU-FAST-Lab/Fast-Drone-250 #43: ego planner die when launched

FPSychotic, 2023-02-17:
roslaunch ego_planner single_run_in_exp.launch
... logging to /home/oem/.ros/log/9f347834-aeb9-11ed-927f-9304ded37c71/roslaunch-ihexa-53043.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ihexa:42275/

SUMMARY
========

PARAMETERS
 * /drone_0_ego_planner_node/bspline/limit_acc: 2.0
 * /drone_0_ego_planner_node/bspline/limit_ratio: 1.1
 * /drone_0_ego_planner_node/bspline/limit_vel: 0.25
 * /drone_0_ego_planner_node/fsm/emergency_time: 1.0
 * /drone_0_ego_planner_node/fsm/fail_safe: True
 * /drone_0_ego_planner_node/fsm/flight_type: 1
 * /drone_0_ego_planner_node/fsm/planning_horizon: 6.0
 * /drone_0_ego_planner_node/fsm/planning_horizon_time: 3.0
 * /drone_0_ego_planner_node/fsm/realworld_experiment: True
 * /drone_0_ego_planner_node/fsm/thresh_no_replan_meter: 1.0
 * /drone_0_ego_planner_node/fsm/thresh_replan_time: 1.0
 * /drone_0_ego_planner_node/fsm/waypoint0_x: 15.0
 * /drone_0_ego_planner_node/fsm/waypoint0_y: 0.0
 * /drone_0_ego_planner_node/fsm/waypoint0_z: 1.0
 * /drone_0_ego_planner_node/fsm/waypoint1_x: 0.0
 * /drone_0_ego_planner_node/fsm/waypoint1_y: 0.0
 * /drone_0_ego_planner_node/fsm/waypoint1_z: 1.0
 * /drone_0_ego_planner_node/fsm/waypoint2_x: 15.0
 * /drone_0_ego_planner_node/fsm/waypoint2_y: 0.0
 * /drone_0_ego_planner_node/fsm/waypoint2_z: 1.0
 * /drone_0_ego_planner_node/fsm/waypoint3_x: 0.0
 * /drone_0_ego_planner_node/fsm/waypoint3_y: 0.0
 * /drone_0_ego_planner_node/fsm/waypoint3_z: 1.0
 * /drone_0_ego_planner_node/fsm/waypoint4_x: 15.0
 * /drone_0_ego_planner_node/fsm/waypoint4_y: 0.0
 * /drone_0_ego_planner_node/fsm/waypoint4_z: 1.0
 * /drone_0_ego_planner_node/fsm/waypoint_num: 1
 * /drone_0_ego_planner_node/grid_map/cx: 162.2635345458984
 * /drone_0_ego_planner_node/grid_map/cy: 117.3401031494141
 * /drone_0_ego_planner_node/grid_map/depth_filter_margin: 2
 * /drone_0_ego_planner_node/grid_map/depth_filter_maxdist: 5.0
 * /drone_0_ego_planner_node/grid_map/depth_filter_mindist: 0.2
 * /drone_0_ego_planner_node/grid_map/depth_filter_tolerance: 0.15
 * /drone_0_ego_planner_node/grid_map/frame_id: world
 * /drone_0_ego_planner_node/grid_map/fx: 307.8522033691406
 * /drone_0_ego_planner_node/grid_map/fy: 310.60693359375
 * /drone_0_ego_planner_node/grid_map/ground_height: -0.01
 * /drone_0_ego_planner_node/grid_map/k_depth_scaling_factor: 1000.0
 * /drone_0_ego_planner_node/grid_map/local_map_margin: 10
 * /drone_0_ego_planner_node/grid_map/local_update_range_x: 5.5
 * /drone_0_ego_planner_node/grid_map/local_update_range_y: 5.5
 * /drone_0_ego_planner_node/grid_map/local_update_range_z: 4.5
 * /drone_0_ego_planner_node/grid_map/map_size_x: 50
 * /drone_0_ego_planner_node/grid_map/map_size_y: 50
 * /drone_0_ego_planner_node/grid_map/map_size_z: 5.0
 * /drone_0_ego_planner_node/grid_map/max_ray_length: 5.0
 * /drone_0_ego_planner_node/grid_map/min_ray_length: 0.3
 * /drone_0_ego_planner_node/grid_map/obstacles_inflation: 0.5
 * /drone_0_ego_planner_node/grid_map/p_hit: 0.65
 * /drone_0_ego_planner_node/grid_map/p_max: 0.9
 * /drone_0_ego_planner_node/grid_map/p_min: 0.12
 * /drone_0_ego_planner_node/grid_map/p_miss: 0.35
 * /drone_0_ego_planner_node/grid_map/p_occ: 0.8
 * /drone_0_ego_planner_node/grid_map/pose_type: 2
 * /drone_0_ego_planner_node/grid_map/resolution: 0.2
 * /drone_0_ego_planner_node/grid_map/show_occ_time: False
 * /drone_0_ego_planner_node/grid_map/skip_pixel: 2
 * /drone_0_ego_planner_node/grid_map/use_depth_filter: True
 * /drone_0_ego_planner_node/grid_map/visualization_truncate_height: 1.8
 * /drone_0_ego_planner_node/manager/control_points_distance: 0.4
 * /drone_0_ego_planner_node/manager/drone_id: 0
 * /drone_0_ego_planner_node/manager/feasibility_tolerance: 0.05
 * /drone_0_ego_planner_node/manager/max_acc: 2.0
 * /drone_0_ego_planner_node/manager/max_jerk: 4.0
 * /drone_0_ego_planner_node/manager/max_vel: 0.25
 * /drone_0_ego_planner_node/manager/planning_horizon: 6.0
 * /drone_0_ego_planner_node/manager/use_distinctive_trajs: False
 * /drone_0_ego_planner_node/optimization/dist0: 0.5
 * /drone_0_ego_planner_node/optimization/lambda_collision: 0.5
 * /drone_0_ego_planner_node/optimization/lambda_feasibility: 0.1
 * /drone_0_ego_planner_node/optimization/lambda_fitness: 1.0
 * /drone_0_ego_planner_node/optimization/lambda_smooth: 1.0
 * /drone_0_ego_planner_node/optimization/max_acc: 2.0
 * /drone_0_ego_planner_node/optimization/max_vel: 0.25
 * /drone_0_ego_planner_node/optimization/swarm_clearance: 0.5
 * /drone_0_ego_planner_node/prediction/lambda: 1.0
 * /drone_0_ego_planner_node/prediction/obj_num: 10
 * /drone_0_ego_planner_node/prediction/predict_rate: 1.0
 * /drone_0_traj_server/traj_server/time_forward: 1.0
 * /rosdistro: noetic
 * /rosversion: 1.15.15

NODES
  /
    drone_0_ego_planner_node (ego_planner/ego_planner_node)
    drone_0_traj_server (ego_planner/traj_server)

ROS_MASTER_URI=http://localhost:11311

process[drone_0_ego_planner_node-1]: started with pid [53178]
process[drone_0_traj_server-2]: started with pid [53179]
hit: 0.619039
miss: -0.619039
min log: -1.99243
max: 2.19722
thresh log: 1.38629
[FSM]: from INIT to WAIT_TARGET
[ WARN] [1676641698.037088225]: [Traj server]: ready.
[drone_0_ego_planner_node-1] process has died [pid 53178, exit code -11, cmd /home/oem/ihexa_ws/devel/lib/ego_planner/ego_planner_node ~odom_world:=/vins_fusion/imu_propagate ~planning/bspline:=/drone_0_planning/bspline ~planning/data_display:=/drone_0_planning/data_display ~planning/broadcast_bspline_from_planner:=/broadcast_bspline ~planning/broadcast_bspline_to_planner:=/broadcast_bspline ~grid_map/odom:=/vins_fusion/imu_propagate ~grid_map/cloud:=nouse2 ~grid_map/pose:=nouse1 ~grid_map/depth:=/camera/depth/image_rect_raw __name:=drone_0_ego_planner_node __log:=/home/oem/.ros/log/9f347834-aeb9-11ed-927f-9304ded37c71/drone_0_ego_planner_node-1.log].
log file: /home/oem/.ros/log/9f347834-aeb9-11ed-927f-9304ded37c71/drone_0_ego_planner_node-1*.log
^C[drone_0_traj_server-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

FPSychotic, 2023-02-18:
I noted if I first launch the node and later vins, it will not crash immediately, will survive a time, and depending the settings , sometimes will last quite more, could CPU load, or IMU with bad sychro ?

2500035435, 2023-03-29:
I also met the same problem. Have you found the solution？

PanNeng-ZJU, 2023-04-12:
The mostly likely reason is that your VIO flies away, and the position is beyond the range of the map size set up of the planner, and it dies.

closed 2023-04-12 by PanNeng-ZJU