ZJU-FAST-Lab/Fast-Drone-250 #53: px4ctrl yaw bug

Poaos, 2023-04-08:
您好，我最近在使用px4ctrl控制四旋翼时遇到了问题:
目前采用t265定位，px4ctrl订阅 /mavros/local_position/odom和/mavros/imu/data，只不过他们都频率都没有高于100hz。不知道是否影响。

px4ctrl控制无人机自动起飞并悬停后，遥控器控制四旋翼的油门，roll，pitch这些都正常，但是控制yaw时，如果yaw角度变化大于100度（大概），无人机就会开始圆周运动，并且是半径越转越大那种（最后会失速往一个方向炸鸡）。config文件中的参数我基本上都已经对应修改过了，那个kp kv也尝试改小过，都是一样的效果。当然，如果我在上述情况下立马切出offboard，飞机就能够一下子稳定住。（过程中/mavros/local_position/odom还是正常的，没有飘）

这个问题是我在飞egoplanner的时候，先设定目标点（3，0，1），正常飞过去，再设定（2，0，1），无人机转向后就会直接往某个方向炸鸡，多次排查后发现的。我目前判断问题应该是px4ctrl本身的，因为纯t265的时候，使用px4自己的位置控制没有出现过这些情况。不知道您是否遇到过这种情况，或者有大概的问题解决思路?

PanNeng-ZJU, 2023-04-12:
t265的速度反馈有问题，如果要用它做里程计的话，需要融合imu数据，你这种类似情况我也遇到过。

EhrazImam, 2023-04-13:
Hi @PanNeng-ZJU even i faced the exact same problem and i am using openvins and vin-fusion where the pixhawk imu and the problem lies same.

PanNeng-ZJU, 2023-04-13:
@EhrazImam @Poaos 问题在于你们使用的里程计的速度反馈有问题，比较正确的解决办法是，将里程计与IMU进行ekf融合，获得一个相对正确的速度估计；相对妥协的办法是，直接在px4ctrl中速度误差反馈的位置，直接把它关掉，但这会导致位置跟踪误差变大

Ding-Guo, 2023-04-20:
我用仿真软件调px_ctrl也遇到了一样的问题

Ding-Guo, 2023-04-24:
我在仿真里面已经解决了我遇到的问题，问题确实如大佬所说是速度反馈出现了问题，对于我的情况是因为反馈的速度的坐标系是相对于机体坐标系，需要对速度做一个旋转变换。