ZJU-FAST-Lab/Fast-Drone-250 #55: 无人机自动起飞失败

Nikanbaturu, 2023-04-09:
按照教程，完成VINS标定后进行自动起飞悬停实验。下发命令后，无人机在自动起飞过程中，一开始起飞到了一个很低的位置，观察来看好像是无人机因为过中飞不起来，随后无人机急剧加速失控，并上升到了很高的位置，但echo /vins_fusion/imu_propogate vins并没有漂，请问这是什么情况导致的？
Ps：
    1.  我的无人机飞控选用的是雷迅V5+
    2.  由于使用的飞控不一样，并且我自行安装了防护圈，飞机重量比视频教程中的要大，约重1.5kg
    3. 起飞后悬停高度设置成了 0.6m，而不是默认的1.0m

PanNeng-ZJU, 2023-04-12:
建议尝试加高config文件里的hover_percentage值到你的实际悬停油门再测试

eleboss, 2023-06-07:
hi pan博您好，
我也遇到了类似的问题，我的整机是S500的框架，起飞的hover percent根据遥控器上油门的数值设置到大概35%左右，无人机缓慢起飞到设定的0.6m时候能够稳定在那个位置，然后过一会就会突然加速冲刺，向上飞，每次出现这种情况我都是直接切手动降落，为什么会这样呢？

eleboss, 2023-06-07:
update，
似乎无人机并没办法稳定到固定的高度，我的无人机晃得很厉害，在设定点附近震荡

eleboss, 2023-06-08:
update,
我尝试用比较小的位置p，和比较大的速度p，貌似都没办法很好的定住。整个飞机就是在高度位置附近震荡，是不是控制器没有微分项导致的呢？

PanNeng-ZJU, 2023-06-08:
请查看一下/mavros/imu/data话题的数据，因为你更换了机架，如果IMU的震动噪声很大，VIO是会很不稳定，从而导致你现在的问题，一般在噪声在+-2以内比较理想 @eleboss

eleboss, 2023-06-08:
@PanNeng-ZJU 感谢回复。我用工具查看了一下飞机在手动模式下的IMU噪声情况，我手动打开螺旋桨，然后把油门推到35%（起飞临界），然后记录了IMU的采样结果。这样一个噪声情况算比较严重的吗？
[image]

eleboss, 2023-06-09:
@PanNeng-ZJU  感谢回复，我更换了一套新的动力系统，噪声能到+-3左右。正在测试，似乎效果明显好转。不过可能是因为控制器PD参数的原因，每次起飞有些许超调，定点依旧有困难。pan博有在大轴距调试参数的经验吗？或者有什么可以分享的经验吗？

eleboss, 2023-06-12:
update,
不是噪声的问题，没用dshot，导致控制缓慢。改了dshot之后就好了。

closed 2023-06-13 by PanNeng-ZJU