ZJU-FAST-Lab/Fast-Drone-250 #60: 关于实际飞行时数据转换的问题

cheslee-z, 2023-05-31:
你好，请问vins-fusion输出的位姿是怎么转换为飞控所需要的定位信息的，我没有找到对应的topic映射关系；此外，d435i相机是如何实现同时输出双目图像和深度图像的呢，我编译该项目的源码后启动realsense_ros节点，用rqt_image_view查看双目图像都具有光点。我知道ego-planner改进了realsense-ros驱动，但是在issue里也提到那个版本过时了，那这个仓库下用的就是原始的版本吗

2500035435, 2023-06-06:
双目图像有光点应该是结构光没关掉吧？此外，d435i的双目和深度图像输出可以修改rs_camera.launch文件，
"enable_depth"设为true输出深度图像，"enable_infra1"和"enable_infra2"设为true输出双目话题。

cheslee-z, 2023-06-06:
> 双目图像有光点应该是结构光没关掉吧？此外，d435i的双目和深度图像输出可以修改rs_camera.launch文件， "enable_depth"设为true输出深度图像，"enable_infra1"和"enable_infra2"设为true输出双目话题。

直接把结构光关掉会被深度图的成像质量有影响吧

Joosoo1, 2023-06-21:
最好还是关掉，要不然vins会很飘

cheslee-z, 2023-06-21:
> 最好还是关掉，要不然vins会很飘

我现在已经关掉了，但是规划必须得用深度图，这怎么办呢

Joosoo1, 2023-06-21:
不用结构光的话，在环境中多布置一点特征点比较多的物品，依靠双目输出的深度图用来规划勉强够用，飞ego是没问题的

FaboNo, 2023-09-16:
@cheslee-z @Joosoo1 Can you tell me how did you turn off the emitter (lumière structurée) ?
你能告诉我你是如何关闭发射器（lumière structurée）的吗？

Joosoo1, 2024-04-08:
> @cheslee-z @Joosoo1 Can you tell me how did you turn off the emitter (lumière structurée) ? 你能告诉我你是如何关闭发射器（lumière structurée）的吗？

我是直接用黑胶带遮住结构光的