ZJU-FAST-Lab/Fast-Drone-250 #63: record.sh 文件中最后一行是否多了一个“\” ?

Biude, 2023-06-16:
[ INFO] [1686912190.511128333]: Subscribing to /camera/infra1/image_rect_raw
[ INFO] [1686912190.512702355]: Subscribing to /drone_0_ego_planner_node/goal_point
[ INFO] [1686912190.513699049]: Subscribing to /drone_0_ego_planner_node/grid_map/occupancy_inflate
[ INFO] [1686912190.514698569]: Subscribing to /drone_0_ego_planner_node/init_list
[ INFO] [1686912190.516005986]: Subscribing to /drone_0_ego_planner_node/optimal_list
[ INFO] [1686912190.517788915]: Subscribing to /drone_0_odom_visualization/path
[ INFO] [1686912190.518836154]: Subscribing to /drone_0_odom_visualization/robot
[ INFO] [1686912190.519707753]: Subscribing to /ego_planner_node/a_star_list
[ INFO] [1686912190.520515342]: Subscribing to /ego_planner_node/global_list
[ INFO] [1686912190.521363340]: Subscribing to /position_cmd
[ INFO] [1686912190.522285830]: Subscribing to /vins_fusion/odometry
[ INFO] [1686912190.523258407]: Subscribing to /vins_fusion/path
[ INFO] [1686912190.524125354]: Subscribing to \
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Character [\] is not valid as the first character in Graph Resource Name [\].  Valid characters are a-z, A-Z, / and in some cases ~.

rosbag record --tcpnodelay \
...
/camera/infra1/image_rect_raw \
/position_cmd

EGOIST5, 2023-08-08:
是的,最后的那个\去掉就好了

Biude, 2023-08-08:
忘记关了，应该就是这个原因了。影响不大

closed 2023-08-08 by Biude