ZJU-FAST-Lab/Fast-Drone-250 #65: VINS飘一直未找到原因

EGOIST5, 2023-07-26:
使用的无人机是JCV-600，相机为D435i，机载电脑为jetson xavier nx，
已将VINS修改为GPU版本，即使用https://github.com/pjrambo/VINS-Fusion-gpu这里的文件替换原来的vins-fusion文件夹，并且在其中添加了相应的fast-drone-250.yaml、left.yaml、right.yaml并替换了原来的vins_rviz_config.rviz，并在fast-drone-250.yaml增加了相应的使用GPU参数，其余未做替换。并仿造ego_planner_swarm项目中将相应的CMakeList.txt内容修改set(ENABLE_CUDA true)及 set(CUDA_NVCC_FLAGS 
      -gencode arch=compute_72,code=sm_72;
    ) 
结构光也已经关闭。
但是无人机即使放在远处，VINS显示的position也将不断增加，seq达到六七千的时候x和y可能达到几十，甚至上百，一直未找到相应的原因，之后按照相应的操作去替换相应的矩阵参数，也未得到改善，并且这几次标定发现vins_output/extrinsic_parameter.txt里的内容一直不变。

otis-XJY, 2023-11-10:
我也使用同样的机载电脑和相机，请问是如何解决的呢

2874160799, 2024-07-15:
请问如何解决的呢？

EGOIST5, 2024-07-15:
我当时是矩阵参数不够准确，替换到更准确的矩阵参数后，就不会飘了

2874160799, 2024-07-15:
> 我当时是矩阵参数不够准确，替换到更准确的矩阵参数后，就不会飘了

是外参和内参的标定矩阵吗

EGOIST5, 2024-07-15:
是的