ZJU-FAST-Lab/Fast-Drone-250 #66: 6c mini 提高IMU发布频率 原命令失效

Zhangyucong0210, 2023-08-10:
[第四章第二步](https://github.com/ZJU-FAST-Lab/Fast-Drone-250#%E7%AC%AC%E5%9B%9B%E7%AB%A0%E9%A3%9E%E6%8E%A7%E8%AE%BE%E7%BD%AE%E4%B8%8E%E8%AF%95%E9%A3%9E)中，写入的命令应该加入一行，最后文件内容应该为
```bash
mavlink start -d /dev/ttyACM0 -b 921600
mavlink stream -d /dev/ttyACM0 -s ATTITUDE_QUATERNION -r 200
mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200
```

FRO4TEN, 2023-08-28:
> 

同为6cmini 依次更改后仍无法提高imu频率，需要在mavros运行后执行
rosrun mavros mavcmd long 511 105 5000 0 0 0 0 0 & sleep 1;
rosrun mavros mavcmd long 511 31 5000 0 0 0 0 0 & sleep 1;
才可以将发布频率提高至170hz左右
p.s. 固件版本为1.13.3

enord-co, 2023-08-29:
> > > [第四章第二步](https://github.com/ZJU-FAST-Lab/Fast-Drone-250#%E7%AC%AC%E5%9B%9B%E7%AB%A0%E9%A3%9E%E6%8E%A7%E8%AE%BE%E7%BD%AE%E4%B8%8E%E8%AF%95%E9%A3%9E)中，写入的命令应该加入一行，最后文件内容应该为
> > > ```shell
> > > mavlink start -d /dev/ttyACM0 -b 921600
> > > mavlink stream -d /dev/ttyACM0 -s ATTITUDE_QUATERNION -r 200
> > > mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200mavlink 启动 -d /dev/ttyACM0 -b 921600
> > > mavlink 流 -d /dev/ttyACM0 -s ATTITUDE_QUATERNION -r 200
> > > mavlink流-d /dev/ttyACM0 -s HIGHRES_IMU -r 200
> > > ```
> > 
> > 
> > 同为6cmini 依次更改后仍无法提高imu频率，需要在mavros运行后执行 rosrun mavros mavcmd long 511 105 5000 0 0 0 0 0 & sleep 1; rosrun mavros mavcmd long 511 31 5000 0 0 0 0 0 & sleep 1; 才可以将发布频率提高至170hz左右 p.s. 固件版本为1.13.3
> 
> 也可以直接修改rspx4脚本部分内容,将IMU提高到200Hz，否则Vins偶尔会出现Wait for imu 的问题
> 
> ```
> roslaunch mavros px4.launch &sleep 10;
> rosrun mavros mavcmd long 511 105 4550 0 0 0 0 0 &sleep 1;
> echo "/mavros/imu/data_raw to 200Hz"
> rosrun mavros mavcmd long 511 31  4550 0 0 0 0 0 &sleep 1;
> echo "/mavros/imu/data to 200Hz"
> ```

Dear @Zhangyucong0210 even i am trying to run with 6c mini, Was it working fine with the ego planner

FRO4TEN, 2023-09-02:
> > > > [第四章第二步](https://github.com/ZJU-FAST-Lab/Fast-Drone-250#%E7%AC%AC%E5%9B%9B%E7%AB%A0%E9%A3%9E%E6%8E%A7%E8%AE%BE%E7%BD%AE%E4%B8%8E%E8%AF%95%E9%A3%9E)中，写入的命令应该加入一行，最后文件内容应该为
> > > > ```shell
> > > > mavlink start -d /dev/ttyACM0 -b 921600
> > > > mavlink stream -d /dev/ttyACM0 -s ATTITUDE_QUATERNION -r 200
> > > > mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200mavlink 启动 -d /dev/ttyACM0 -b 921600
> > > > mavlink 流 -d /dev/ttyACM0 -s ATTITUDE_QUATERNION -r 200
> > > > mavlink流-d /dev/ttyACM0 -s HIGHRES_IMU -r 200
> > > > ```
> > > 
> > > 
> > > 同为6cmini 依次更改后仍无法提高imu频率，需要在mavros运行后执行 rosrun mavros mavcmd long 511 105 5000 0 0 0 0 0 & sleep 1; rosrun mavros mavcmd long 511 31 5000 0 0 0 0 0 & sleep 1; 才可以将发布频率提高至170hz左右 p.s. 固件版本为1.13.3
> > 
> > 
> > 也可以直接修改rspx4脚本部分内容,将IMU提高到200Hz，否则Vins偶尔会出现Wait for imu 的问题
> > ```
> > roslaunch mavros px4.launch &sleep 10;
> > rosrun mavros mavcmd long 511 105 4550 0 0 0 0 0 &sleep 1;
> > echo "/mavros/imu/data_raw to 200Hz"
> > rosrun mavros mavcmd long 511 31  4550 0 0 0 0 0 &sleep 1;
> > echo "/mavros/imu/data to 200Hz"
> > ```
> 
> 亲爱的@Zhangyucong0210 @FRO4TEN ，即使我正在尝试使用 6c mini 运行，它与 ego planner 一起工作正常吗？

可以正常使用

enord-co, 2023-09-02:
@FRO4TEN 感谢您的回复，您在 6c mini 上尝试过此存储库吗

FRO4TEN, 2023-09-09:
> @FRO4TEN 感谢您的回复，您在 6c mini 上尝试过此存储库吗

我目前使用的就是6cmini