ZJU-FAST-Lab/Fast-Drone-250 #74: out_path不飘，但是/vins_fusion/imu_propagate漂移明显

otis-XJY, 2023-12-07:
[image]
[image]
运行后发现vins输出的out_path不飘，但是/vins_fusion/imu_propagate中的位置漂移明显
我不知道是什么原因导致的？
我发现项目中并没有使用loop_fusion，可以请问一下具体的原因吗？
硬件：D435i+NX（fusin-gpu）

otis-XJY, 2023-12-10:
我还很好奇为什么用imu_propagate作为odom而不是vins输出的odom

mosongnan, 2023-12-28:
我也是这个问题，大佬你解决了吗？

otis-XJY, 2023-12-28:
就是定位失败出现飘逸，感觉是VINS-GPU的问题或者是哪个数据的帧率没上去，之前的环境有点乱了，没有敢刷机重做。

> 我还很好奇为什么用imu_propagate作为odom而不是vins输出的odom

这个问题猜测是因为vins输出的odom的帧率不够

otis-XJY, 2023-12-28:
> 我也是这个问题，大佬你解决了吗？

你可以分享一下你的具体环境吗，我们可以交叉比对一下

mosongnan, 2023-12-28:
> > 我也是这个问题，大佬你解决了吗？
> 
> 你可以分享一下你的具体环境吗，我们可以交叉比对一下

我用的是雷迅v5的飞控jetson tx2nx和D435i，imu数据是飞控的。跑的vins-fusion

otis-XJY, 2023-12-28:
我用的是雷迅v5+的飞控，jetson nx和D435i,跑的vins-fusion-gpu。opencv4.2.0用的GPU版本。你有确定imu、图像等数据的刷新率吗

mosongnan, 2023-12-28:
> 我用的是雷迅v5+的飞控，jetson nx和D435i,跑的vins-fusion-gpu。opencv4.2.0用的GPU版本。你有确定imu、图像等数据的刷新率吗

是按教程来的，imu200hz左右，图像我也不清楚，opencv4.6.0

otis-XJY, 2023-12-28:
我最近有点忙，两周后如果您没解决希望可以和您进一步探讨

mosongnan, 2023-12-28:
> 我最近有点忙，两周后如果您没解决希望可以和您进一步探讨

好的，感谢

ffxunmeng, 2024-04-19:
> 我最近有点忙，两周后如果您没解决希望可以和您进一步探讨

想问一下大佬们，这个问题解决了吗

mosongnan, 2024-04-19:
> > 我最近有点忙，两周后如果您没解决希望可以和您进一步探讨
> 
> 想问一下大佬们，这个问题解决了吗

换成nuc以后就没这个问题了，怀疑是算力问题。之前用的是jetson tx2。

Biangsama, 2024-11-09:
想问下大佬们解决了这个问题了吗，我现在也是卡在这一步了

Biangsama, 2024-11-09:
那可以用path的数据作为位置坐标呢

Biangsama, 2024-11-09:
> > > 我最近有点忙，两周后如果您没解决希望可以和您进一步探讨
> > 
> > 
> > 想问一下大佬们，这个问题解决了吗
> 
> 换成nuc以后就没这个问题了，怀疑是算力问题。之前用的是jetson tx2。

那后续还有用JETSON NX解决问题吗

Biangsama, 2024-11-09:
> 我用的是雷迅v5+的飞控，jetson nx和D435i,跑的vins-fusion-gpu。opencv4.2.0用的GPU版本。你有确定imu、图像等数据的刷新率吗

大佬你后续怎么解决的呀