ZJU-FAST-Lab/Fast-Drone-250 #81: 编译时出现问题

xiaoyangjun233, 2024-05-06:
make[2]: *** [realflight_modules/VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/build.make:76：realflight_modules/VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/calib/CameraCalibration.cc.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:2278：realflight_modules/VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/all] 错误 2

报错如上望大佬解救

pinyinjj, 2024-05-30:
同样报错，请问怎么解决

XXLiu-HNU, 2024-06-10:
我遇到了相同的步骤，后来解决了。
解决方法是通过按照readme中的步骤，将**“realsense驱动安装”**和**“****安装ceres与glog与ddyanmic-reconfigure”**部分编译成功，之后在进行**“下载ego-planner源码并编译”**。
亲测对我自己可用，可以检查一下上边步骤是否完成

XXLiu-HNU, 2024-06-11:
先把这部分内容编译成功，再编译代码。我的就是这样解决的

XXLiu-HNU, 2024-06-11:
[image]
先把这部分内容编译完成，再编译后边的内容

xiaoyangjun233, 2024-12-02:
这里可以参考作者后续有在issue中提到针对不同opencv版本与c++环境如何更改。至少我的问题在更改相应函数后即可编译，如有后来的朋友们仍有问题，也希望能与我这个新手一起交流。

closed 2024-12-02 by xiaoyangjun233