ZJU-FAST-Lab/Fast-Drone-250 #84: 无人机缓慢起飞到设定的1m时候能够稳定在那个位置，然后过一会就会突然加速冲刺，向上飞

xj1429796845, 2024-07-03:
您好，我的机架是380机型，飞控是px4  2.4.8  ，起飞的hover percent根据遥控器上油门的数值设置到0.39左右，无人机缓慢起飞到设定的1m时候能够稳定在那个位置，然后过一会就会突然加速冲刺，向上飞，每次出现这种情况我都是直接切手动降落，为什么会这样呢？

LUCKEYgirl, 2024-07-06:
我也是，有解决吗？

xj1429796845, 2024-07-07:
> 我也是，有解决吗？

暂时还没有，我先换了个飞控，下周再试飞一下，如果您解决了，能否告知一下 解决方案

LUCKEYgirl, 2024-07-07:
我解决了，是遥控器油门通道反向了！！！！您看一下是不是3通道的油门上拨，舵量变小，不然就不对，你可以仔细看一下视频，然后如果还是不行那就是  hover_percentage: 0.30  # Thrust percentage in Stabilize/Arco mode # *太大了，慢慢调小到刚刚能够起飞的感觉，最好把飞机用绳子拴住直留能够飞1m，然后飞过头就急停，机臂下面垫海绵

LUCKEYgirl, 2024-07-07:
不是飞控的问题

LUCKEYgirl, 2024-07-07:
我刚刚解决了，是遥控器油门通道反向了！！！！您看一下是不是3通道的油门上拨，舵量变小，不然就不对，你可以仔细看一下视频，然后如果还是不行那就是 hover_percentage: 0.30 # Thrust percentage in Stabilize/Arco mode # *太大了，慢慢调小到刚刚能够起飞的感觉，最好把飞机用绳子拴住直留能够飞1m，然后飞过头就急停，机臂下面垫海绵，我们之前也怀疑是飞控，目前不是飞控的问题

xj1429796845, 2024-07-07:
> 我刚刚解决了，是遥控器油门通道反向了！！！！您看一下是不是3通道的油门上拨，舵量变小，不然就不对，你可以仔细看一下视频，然后如果还是不行那就是 hover_percentage: 0.30 # Thrust percentage in Stabilize/Arco mode # *太大了，慢慢调小到刚刚能够起飞的感觉，最好把飞机用绳子拴住直留能够飞1m，然后飞过头就急停，机臂下面垫海绵，我们之前也怀疑是飞控，目前不是飞控的问题
> […](#)

我刚看了我的3通道是正确的，向上拨舵量 是变小的，那我看下是否是 hover_percentage 值太大的原因，感谢！！

LUCKEYgirl, 2024-07-07:
向上拨舵量是从下面逐渐变小的，不是上面逐渐变小的！一定注意！

LUCKEYgirl, 2024-07-07:
这种就是错的，需要改反相的，我们炸鸡两次了，一定要注意

xj1429796845, 2024-07-07:
> 这种就是错的，需要改反相的，我们炸鸡两次了，一定要注意
> […](#)

我看浙大写的是，，，看echo出来的消息里哪项在缓慢变化（这项就是油门通道值），并观察它是不是由小变大
如果是由小变大，则不需要修改throttle的rc_reverse，反之改为true，，虽然我看打印rostopic echo /mavros/rc/in出来的是 由小变大，，，，，那会儿我看的是遥控器上的舵量值向上拨是由大变小，所以我直接将throttle改成true 对吧，

xj1429796845, 2024-07-07:
> 这种就是错的，需要改反相的，我们炸鸡两次了，一定要注意
> […](#)

我的遥控器AT9S PRO，油门再最低的时候，我看遥控器上的舵量值居然是100。。。。。。。

xj1429796845, 2024-07-07:
> 向上拨舵量是从下面逐渐变小的，不是上面逐渐变小的！一定注意！
> […](#)

不好意思 我刚看懂了，，原先我的油门，从下往上面打的时候 ，，，那个舵量值 ，是从上面逐渐变小， 也就是你说的反了，舵量值需要从下面，，逐渐变小

LUCKEYgirl, 2024-07-07:
对的，那样是错误的

xj1429796845, 2024-07-08:
> 对的，那样是错误的
> […](#)

我设置反向之后，自稳模式，都不能解锁电机了

xj1429796845, 2024-07-08:
> 我设置原先是反相改成正相之后，油门往上打的时候，舵量值从下面 逐渐减小，我重新校准遥控器 就可以解锁了

xj1429796845, 2024-07-08:
> 对的，那样是错误的
> […](#)

我设置原先是反相改成正相之后，油门往上打的时候，舵量值从下面 逐渐减小，我重新校准遥控器 就可以解锁了

xj1429796845, 2024-07-08:
> 对的，那样是错误的
> […](#)

您改了之后 ，查看rostopic echo /mavros/rc/in出来的是 由大变小没，有没有改throttle，，，

xj1429796845, 2024-07-08:
> 这种就是错的，需要改反相的，我们炸鸡两次了，一定要注意
> […](#)

我已经解决了，我是直接把throttle改成了true ，然后我换了飞控，现在起飞稳稳的，感谢您，非常感谢您提供思路

LUCKEYgirl, 2024-07-08:
我们没有改throttle这个，但是您改了能够跑成功，那就也ok，我们没改，然后重新校准遥控器 ，就可以顺利起飞

xj1429796845, 2024-07-09:
> 我们没有改throttle这个，但是您改了能够跑成功，那就也ok，我们没改，然后重新校准遥控器 ，就可以顺利起飞

应该这两种方式二选一都可以，我跑了ego-planner了，再也不往天上冲了，稳稳的

closed 2024-11-11 by PanNeng-ZJU

XXLiu-HNU, 2025-04-04:
遇到了同样的问题，再咨询一下大佬们。

首先我的遥控器油门规则如下:

通过拨动油门位置查看mavros rc in 话题的变化

1. 当油门在**最下边**的时候输出为**999**

2. 当油门在**中间**的时候输出为**1500**，

3. 当油门在**最上边**的时候输出为**1999**；

同时我也去阅读了PX4CtrlFSM.cpp 文件，发现当无人机从自动起飞进入悬停模式之后，会根据下边这个函数计算期望位置

```
void PX4CtrlFSM::set_hov_with_rc()
{
	ros::Time now = ros::Time::now();
	double delta_t = (now - last_set_hover_pose_time).toSec();
	last_set_hover_pose_time = now;

	hover_pose(0) += rc_data.ch[1] * param.max_manual_vel * delta_t * (param.rc_reverse.pitch ? 1 : -1);
	hover_pose(1) += rc_data.ch[0] * param.max_manual_vel * delta_t * (param.rc_reverse.roll ? 1 : -1);
        // 这一行对应的应该是高度的值，根据遥控器杆量来生成期望的位置。
	hover_pose(2) += rc_data.ch[2] * param.max_manual_vel * delta_t * (param.rc_reverse.throttle ? 1 : -1);
	hover_pose(3) += rc_data.ch[3] * param.max_manual_vel * delta_t * (param.rc_reverse.yaw ? 1 : -1);

	if (hover_pose(2) < -0.3)
		hover_pose(2) = -0.3;
```

因此我有个疑问是，即使油门反向了，此时输出仍是1500，根据遥控器逻辑
```
void RC_Data_t::feed(mavros_msgs::RCInConstPtr pMsg)
{
    msg = *pMsg;
    rcv_stamp = ros::Time::now();

    for (int i = 0; i < 4; i++)
    {
        // 当油门在中位的时候，ch[2]应该是0
        ch[i] = ((double)msg.channels[i] - 1500.0) / 500.0;
        if (ch[i] > DEAD_ZONE)
            ch[i] = (ch[i] - DEAD_ZONE) / (1 - DEAD_ZONE);
        else if (ch[i] < -DEAD_ZONE)
            ch[i] = (ch[i] + DEAD_ZONE) / (1 - DEAD_ZONE);
        else
            ch[i] = 0.0;
    }
```
此时ch[2]应该已知是0，为什么无人机会不断往天上飞。因此不太理解为什么会出现这个问题。

chenfch3, 2025-05-02:
> 遇到了同样的问题，再咨询一下大佬们。
>
> 首先我的遥控器油门规则如下:
>
> 通过拨动油门位置查看mavros rc in 话题的变化
>
> 1. 当油门在**最下边**的时候输出为**999**
> 2. 当油门在**中间**的时候输出为**1500**，
> 3. 当油门在**最上边**的时候输出为**1999**；
>
> 同时我也去阅读了PX4CtrlFSM.cpp [...]

> 	hover_pose(2) += rc_data.ch[2] * param.max_manual_vel * delta_t * (param.rc_reverse.throttle ? 1 : -1);
You shoule set param.rc_reverse.throttle to true and the  (param.rc_reverse.throttle ? 1 : -1) will set to 1.