ZJU-FAST-Lab/Fast-Drone-250 #96: 自动起飞时无人机向左飘/无人机实际位置和规划出来的轨迹存在误差

XXLiu-HNU, 2024-10-29:
### **背景：**  
采用的机架为450机架，飞控为Cuav X7+ ，动力套件正常，在RTK模式下定点模式稳定
修改 `ctrl_param_fpv.yaml` 文件中的 `mass`，PID参数 `Kp0 …`从1.5修改为2

### **问题1：** 
执行自动起飞的时候，无人机会向左方平移大概**20cm**的距离，然后上升到1m的距离。应该不是什么大问题，但是出现了问题2，怀疑和问题1类似，附上视频，具体体现在视频的1s-3s区间：
https://github.com/user-attachments/assets/afea5c59-82da-4c89-a78e-7710202974bb

### **问题2：**
不确定是否和 **问题一** 存在关联，通过录包会看的过程，发现无人机的轨迹（绿色细线段）和规划出来的轨迹（红色粗线段）存在一定的偏差，同样是往左偏移大概 **20cm** 的距离， **实际飞行中发现VINS定位未发生漂移情况** 

### 推断可能的原因：

1. 无人机传感器需要校准，因为实验过程中发生了几次摔机，可能传感器需要校准
2. 控制部分参数需要修改，因为使用的飞机和教程中的飞机存在区别

### 希望实现的效果
无人机准确的跟踪生成的轨迹，从而实现正确的避障行为

XXLiu-HNU, 2024-10-29:
通过重新校准飞机的传感器，实现了精准的跟踪

closed 2024-10-29 by XXLiu-HNU

unius589, 2025-01-21:
您好,最近也在复现这个项目.我的飞机也是大轴距, 我现在飞机起飞后会有左右漂移的问题.  现在想调整一下pid参数, 具体思路有点混乱, 想跟你请教一下: 
1. 我看它这个项目只能调整ctrl_param_fpv.yam文件中的pid参数, 我的理解是这个跟踪控制是在程序中controller.cpp实现的, 底层飞控px4的唯一作用就是执行上层的控制指令(话题为/mavros/setpoint_raw/attitude), 而跟和px4自身的pid参数是无关, 不知这样理解对不对?
2.  您提到的无人机轨迹 和 规划出来的轨迹的对比, 具体是指什么? 是无人机的里程计/odom  和  des.p的信息吗?

谢谢

unius589, 2025-01-21:
@XXLiu-HNU 您好,最近也在复现这个项目.我的飞机也是大轴距, 我现在飞机起飞后会有左右漂移的问题. 现在想调整一下pid参数, 具体思路有点混乱, 想跟你请教一下:

    我看它这个项目只能调整ctrl_param_fpv.yam文件中的pid参数, 我的理解是这个跟踪控制是在程序中controller.cpp实现的, 底层飞控px4的唯一作用就是执行上层的控制指令(话题为/mavros/setpoint_raw/attitude), 而跟和px4自身的pid参数是无关, 不知这样理解对不对?
    您提到的无人机轨迹 和 规划出来的轨迹的对比, 具体是指什么? 是无人机的里程计/odom 和 des.p的信息吗?

谢谢

unius589, 2025-01-21:
@XXLiu-HNU

XXLiu-HNU, 2025-01-21:
> 您好,最近也在复现这个项目.我的飞机也是大轴距, 我现在飞机起飞后会有左右漂移的问题. 现在想调整一下pid参数, 具体思路有点混乱, 想跟你请教一下:
>
> 1. 我看它这个项目只能调整ctrl_param_fpv.yam文件中的pid参数, 我的理解是这个跟踪控制是在程序中controller.cpp实现的, [...]
首先我也不是专门做控制的，这是我自己的一些看法： 
1. 这里的PID是根据期望位置和当前位置的偏差，计算控制量，具体你可以参考[control.cpp](https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/master/src/realflight_modules/px4ctrl/src/controller.cpp#L22)。这个PID参数和飞控类似，不过他给的默认的参数基本上都是可以用的，左右飘一般不是PID参数的问题，我猜测是飞控IMU未增加减震模块或者飞控传感器未校准。 这里的PID并不是和自身PID无关，因为一般飞控可以直接输入位置控制（setpoint/local_pose）这个时候飞控自己就需要进行位置环和姿态环的PID参数控制，相当于这里单独把位置环拿出来，自己计算，然后位置环还是飞控进行控制。
2. 这里是指：规划的轨迹（规划的轨迹） 与 无人机的实际位置（odom），表现在RVIZ中无人机的轨迹（绿色）和规划的路径（红色）不重合
3. 建议你校准一下传感器，然后看看IMU是不是震动太大

XXLiu-HNU, 2025-01-21:
> [@XXLiu-HNU](https://github.com/XXLiu-HNU)

这里有关于控制部分理论的讲解：【无人机系统 - 高飞组超全讲解！2021ROS暑期学校 第五天精剪回放】 https://www.bilibili.com/video/BV1Jq4y1T7QD/?p=3&share_source=copy_web&vd_source=694800f4f1ae99186a15066decbc1bc2

unius589, 2025-02-08:
@XXLiu-HNU 感谢回答！突然有个疑问，按照这个逻辑，其实位置和速度控制也放到PX4底层去做，也就是我们只给飞控发送位置控制指令；这样是否会 位置和速度控制放在上层去做要更好？

unius589, 2025-02-08:
还有，飞控需要执行姿态环（及角速度环）的控制，那么姿态环的PID参数理论上也需要在QGC中进行设置调整的对吧？

XXLiu-HNU, 2025-02-08:
> @XXLiu-HNU 感谢回答！突然有个疑问，按照这个逻辑，其实位置和速度控制也放到PX4底层去做，也就是我们只给飞控发送位置控制指令；这样是否会 位置和速度控制放在上层去做要更好？

这个需要看你的需求，如果无人机的轨迹没有很激进，只用位置控制也是可以的（个人意见），高飞老师在 b 站说想复现他们的项目最好参考他们的控制器逻辑

XXLiu-HNU, 2025-02-08:
> 还有，飞控需要执行姿态环（及角速度环）的控制，那么姿态环的PID参数理论上也需要在QGC中进行设置调整的对吧？

是的，这个项目给的参数是一些比较通用的参数，针对自己的飞机可以自己调整来适配已实现更好的效果