ZJU-FAST-Lab/Fast-Drone-250 #98: IMU problem and automatic takeoff error

yanjifa666, 2024-11-05:
I am using a 6c flight controller. I increased the frequency of the imu through commands. However, when testing automatic takeoff, it still prompts that the frequency is too low and the odometer speed is too high, so the takeoff is restricted.
[image]

PanNeng-ZJU, 2024-11-11:
You should firstly check the IMU frequency by the command "rostopic hz /mavros/imu/data" to see if you have really increased the frequency. As for the high odometry speed, it's obviously that the odometry has broken down, and you should check it first.

yanjifa666, 2024-11-25:
Thank you, it has been solved. But before each takeoff, I have to use the command line to increase the imu frequency. (I use a 6c flight controller)