ZJU-FAST-Lab/Fast-Drone-250 #100: Automatic takeoff failed, but there were no errors reported

zhao6868999, 2024-12-16:
[image]
[image]
[image]
The first takeoff prompt showed a low IMU frequency. After using the mavlink command to increase the frequency, running the takeoff command again resulted in no error messages. Is there anyone who can answer this question?

XXLiu-HNU, 2024-12-18:
1. Check the UAV state in QGC or mavros/state, is armed and offboarded?
2. Make sure the RC's set is right
3. In [PX4CtrlFSM.cpp](https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/master/src/realflight_modules/px4ctrl/src/PX4CtrlFSM.cpp) , you can add ROSINFO for get the FSM's state
4. Something else possible...

zhao6868999, 2024-12-20:
> 1. 检查 QGC 或 mavros/state 中的无人机状态，是否已武装和下机？
> 2. 确保 RC 的集合正确
> 3. 在 [PX4CtrlFSM.cpp](https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/master/src/realflight_modules/px4ctrl/src/PX4CtrlFSM.cpp) 中，您可以添加 ROSINFO 来获取 FSM 的状态
> 4. 其他可能的事情......

Hello, can you explain the third point in detail?

XXLiu-HNU, 2024-12-20:
> > 1. 检查 QGC 或 mavros/state 中的无人机状态，是否已武装和下机？
> > 2. 确保 RC 的集合正确
> > 3. 在 [PX4CtrlFSM.cpp](https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/master/src/realflight_modules/px4ctrl/src/PX4CtrlFSM.cpp) 中，您可以添加 ROSINFO 来获取 FSM 的状态
> > 4. 其他可能的事情......
> 
> Hello, can you explain the third point in detail?
> 

在px4ctrl中，控制器状态机是从munual control 到 auto takeoff 其中触发条件是遥控器设置正确且档位正确，接收到起飞的话题。因为你可以看一下状态机处于什么状态，如果处于手动控制，说明是因为没有成功进入自动起飞模式，如果是在自动起飞，而没有起飞 可能是因为“自动起飞”中有bug，先试着缩小问题的范围