ZJU-FAST-Lab/Fast-Drone-250 #103: 构建简化版的px4ctrl

Jamin-Yang, 2025-01-08:
我想在px4ctrl的基础上进行修改，只使用x-y平面的速度和yaw角的旋转速率进行飞行控制。我可以直接去掉des.a和des.p吗？
 des_acc(0) = Kv(0) * (des.v(0) - odom.v(0)); 
  des_acc(1) = Kv(1) * (des.v(1) - odom.v(1));
  des_acc(2) = 0.0;
  des_acc += Eigen::Vector3d(0, 0, param_.gra);

XXLiu-HNU, 2025-01-11:
这px4ctrl最后用到的话题输出是
```
fsm.ctrl_FCU_pub = nh.advertise<mavros_msgs::AttitudeTarget>("/mavros/setpoint_raw/attitude", 10);
```

简化为速度输出可以使用`/mavros/setpoint_velocity`话题

SriKrishnaKalyan5, 2025-05-15:
你好，有人可以建议我在哪里发布这个 cmd 主题吗？
https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/8ff64274b161bb829653faf5946656658e7e2c9f/src/realflight_modules/px4ctrl/src/px4ctrl_node.cpp#L44

XXLiu-HNU, 2025-05-15:
> 您好，有人可以告诉我这个 CMD 主题在哪里发布吗？
> 
> [Fast-Drone-250/src/realflight_modules/px4ctrl/src/px4ctrl_node.cpp](https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/8ff64274b161bb829653faf5946656658e7e2c9f/src/realflight_modules/px4ctrl/src/px4ctrl_node.cpp#L44)
> 
> Line 44 in [8ff6427](/ZJU-FAST-Lab/Fast-Drone-250/commit/8ff64274b161bb829653faf5946656658e7e2c9f)
> 
>  nh.subscribe<quadrotor_msgs::PositionCommand>("cmd",

https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/8ff64274b161bb829653faf5946656658e7e2c9f/src/planner/plan_manage/src/traj_server.cpp#L241

XXLiu-HNU, 2025-05-15:
> [@XXLiu-HNU](https://github.com/XXLiu-HNU) Yes I see that, but the topic names are different, and from what I understand they should be the same in order to communicate, right?

In the launch file, it uses remap to correspond to the topic name:
https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/8ff64274b161bb829653faf5946656658e7e2c9f/src/realflight_modules/px4ctrl/launch/run_ctrl.launch#L8

SriKrishnaKalyan5, 2025-05-15:
@XXLiu-HNU  Ohhhh Perfect.... Thank you so much for that i missed to go through launch files.... This helps a lot...