ZJU-FAST-Lab/Fast-Drone-250 #108: 可不可以使用MID360 的IMU作为px4ctrl的输入

XXLiu-HNU, 2025-04-04:
目前遇到无人机在自动起飞的时候总是左右晃动，怀疑是飞控自身IMU的减震比较差（NxtPx4 v2）

因此想直接使用MID360雷达的IMU，不知道可不可以

XXLiu-HNU, 2025-04-07:
https://github.com/ZJU-FAST-Lab/ego-planner/issues/112#issuecomment-2581663730

[image]

后续会进行测试，看看效果如何

chenfch3, 2025-04-27:
> [ZJU-FAST-Lab/ego-planner#112 (comment)](https://github.com/ZJU-FAST-Lab/ego-planner/issues/112#issuecomment-2581663730)
> 
> [image]
> 
> 后续会进行测试，看看效果如何

hello, have you tested it? I found a comment [nh.subscribe<sensor_msgs::Imu>("/mavros/imu/data", // Note: do NOT change it to /mavros/imu/data_raw !!!](https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/8ff64274b161bb829653faf5946656658e7e2c9f/src/realflight_modules/px4ctrl/src/px4ctrl_node.cpp#L51). The imu created by livox was raw. How did you fixed it at the end?

XXLiu-HNU, 2025-04-27:
> hello, have you tested it? I found a comment [nh.subscribe<sensor_msgs::Imu>("/mavros/imu/data", // Note: do NOT change it to /mavros/imu/data_raw [...]

实际上我们两种都测试了，效果差不多，不需要额外处理raw的问题。

但是有一个机架雷达的imu震动较大，[hku_mars 作者建议使用飞控的imu作为输入](https://github.com/hku-mars/SUPER-Hardware/issues/6#issuecomment-2823466325)

XXLiu-HNU, 2025-05-15:
更新一下，后边我们还是没用px4ctrl，因为没有找到具体的BUG，飞机也摔坏了好几次！不过借鉴px4ctrl自己写了一个[控制器](https://github.com/HNU-CAT/pid_ctrl)，现在飞的稳稳的！

[测试效果](https://github.com/user-attachments/assets/28321dd7-60d3-4f61-940d-227ff98c5790)

SriKrishnaKalyan5, 2025-05-15:
你好，有人可以建议我在哪里发布这个 cmd 主题吗？
https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/8ff64274b161bb829653faf5946656658e7e2c9f/src/realflight_modules/px4ctrl/src/px4ctrl_node.cpp#L44

SuperLDG, 2025-07-26:
> 实际上我们两种都测试了，效果差不多，不需要额外处理raw的问题。
>
> 但是有一个机架雷达的imu震动较大，[hku_mars 作者建议使用飞控的imu作为输入](https://github.com/hku-mars/SUPER-Hardware/issues/6#issuecomment-2823466325)

大佬您好，想请教您一下，使用mid360实现，目前用的fastlio的odom，也修改了single _run_exp.launch的里程计话题和点云话题，px4ctrl中的里程计话题也修改了，完成这样就能实现用雷达定位完成egoplanner 的避障效果了吗，目前手持雷达运动，egoplanner的规划是会随着变化的，就是没有上无人机，不知道目前的工作可行吗

SuperLDG, 2025-07-26:
以及，您有做过tf转换的工作吗，还是直接使用fastlio输出的odomtry就行了

XXLiu-HNU, 2025-07-26:
> 以及，您有做过tf转换的工作吗，还是直接使用fastlio输出的odomtry就行了

lio可以通过imu差值来实现更高的输出频率，参考vins，一般是提高到imu的频率（200hz），没有做TF变换

quinnym233, 2025-08-25:
[ERROR] [1756119541.847344188]: Ran out of pool, index=92 100 18
[ERROR] [1756119541.847388574]: Unable to handle the initial or end point, force return!
[ERROR] [1756119541.847409566]: a star error, force return!
[ERROR] [1756119541.847451808]: ERROR! the drone is in obstacle. This should not happen.
[ERROR] [1756119541.847474720]: Ran out of pool, index=95 100 15
[ERROR] [1756119541.847491105]: Unable to handle the initial or end point, force return!
[ERROR] [1756119541.847507489]: a star error
iter=5,time(ms)=0.099,rebound.
[ERROR] [1756119541.847558243]: ERROR! the drone is in obstacle. This should not happen.
[ERROR] [1756119541.847579332]: Ran out of pool, index=83 100 27
[ERROR] [1756119541.847596164]: Unable to handle the initial or end point, force return!
[ERROR] [1756119541.847608836]: a star error
iter=5,time(ms)=0.081,rebound.
[ERROR] [1756119541.847642374]: ERROR! the drone is in obstacle. This should not happen.
[ERROR] [1756119541.847660902]: Ran out of pool, index=99 100 12
[ERROR] [1756119541.847675271]: Unable to handle the initial or end point, force return!
[ERROR] [1756119541.847687879]: a star error
iter=5,time(ms)=0.072,rebound.
[ERROR] [1756119541.847720136]: ERROR! the drone is in obstacle. This should not happen.
[ERROR] [1756119541.847738857]: Ran out of pool, index=100 100 12
[ERROR] [1756119541.847751561]: Unable to handle the initial or end point, force return!
[ERROR] [1756119541.847764809]: a star error
iter=5,time(ms)=0.072,rebound.
[ERROR] [1756119541.847798090]: ERROR! the drone is in obstacle. This should not happen.
[ERROR] [1756119541.847845452]: Ran out of pool, index=100 98 11
[ERROR] [1756119541.847874989]: Unable to handle the initial or end point, force return!
[ERROR] [1756119541.847891085]: a star error